nakujaproject/internship2021 #3: Design and fabrication of parachute (N-1)

shohei, 2021-02-27:


dmashy, 2021-03-03:
Thank you Rodney for Uploading a possible parachute on the scrapbox:

I request that you:
Determine 1. the possible supplier and cost of the parachute in Kenya 2. Possible material in case we can buy locally the material and create it in-house or with the aid of a tailor.

Asante

0x6f736f646f, 2021-03-03:
- We discussed with @JeffMboya to fabricate our own parachute. 
- We have listed the possible materials and I will check on them tomorrow at our local supermarket.

shohei, 2021-03-04:
@0x6f736f646f This is noted. Let me know what you found in the supermarket.

dmashy, 2021-03-04:
Noted with thanks.

shohei, 2021-03-04:
FYI: a guideline for sizing parachute
[Newsletter149 Properly Sizing Your Rockets.pdf](https://github.com/nakujaproject/intern/files/6082826/Newsletter149.pdf)

dmashy, 2021-03-04:
Got it, thanks!

0x6f736f646f, 2021-03-23:
Done. ✅

closed 2021-03-23 by 0x6f736f646f