nakujaproject/internship2021 #5: Development of Flight Computer and Electronics (FCE)

shohei, 2021-02-27:
Flight Computer and Electronics (FCE) for N-1 requires the following functions

- [ ] Flight logging using following sensors
  - [ ] Altimeter
  - [ ] Accelerometer
  - [ ] Gyroscope
  - [ ] Others
- [ ] Detection of apogee (maximum altitude)
- [ ] Parachute ejection

0x6f736f646f, 2021-03-11:
For the mechanical part check out [this](https://github.com/nakujaproject/airframev1/blob/main/Design/electronicsbay.stl)

The electronics PCB will be able to be screwed on the avionic bay

For refinement the avionics team should be able to:

- [ ] Place the components on one PCB
- [ ] The back will be an attachment for the battery pack
- [ ] The PCB should have drill holes to be able to secure it correctly to the avionics bay
- [ ] Provide the actual length and width of the  #PCB

0x6f736f646f, 2021-03-23:
I am currently printing the [avionics bay](https://github.com/nakujaproject/airframev1/blob/main/Design/electronicsbay.pdf) where the @nakujaproject/avionics-group will house their components

0x6f736f646f, 2021-03-23:
[image]

Mike-Kimani, 2021-04-12:
@0x6f736f646f    What should be the diameter of the mounting holes on the PCB?

0x6f736f646f, 2021-04-12:
1 mm

closed 2021-05-02 by Mike-Kimani