nakujaproject/internship2021 #16: Redefine N-1 airframe dimension

shohei, 2021-03-01:
- [ ] Adjust diameter depending on the diameter of SRM
- [ ] Adjust diameter/length depending on the N-1 avionics bay

0x6f736f646f, 2021-03-02:
@1998-felix Provide the length and outside diameter of the SRM

0x6f736f646f, 2021-03-02:
#5  @Mike-Kimani Provide length and width of the avionics bay

0x6f736f646f, 2021-03-11:
Tentative the first design is at [here](https://github.com/nakujaproject/airframev1)

Also the open rocket simulation is at [here](https://github.com/nakujaproject/airframe_openrocket)

0x6f736f646f, 2021-03-11:
@dmashy and @shohei please check it out

> Tentative the first design is at [here](https://github.com/nakujaproject/airframev1)
> 
> Also the open rocket simulation is at [here](https://github.com/nakujaproject/airframe_openrocket)

0x6f736f646f, 2021-03-23:
Diameter is adjusted to 50mm

0x6f736f646f, 2021-04-06:
Diameter has been adjusted to 53

closed 2021-04-06 by 0x6f736f646f