nakujaproject/internship2021 #19: Documentation of safety launch sequence

shohei, 2021-03-01:
- [ ] Operation manufal for launch 
  - [ ] Ignition system 
  - [ ] Wind criteria (launch or abort)
  - [ ] Staff arrangement (Who will be doing what)
  - [ ] Communication (tranceiver, etc.)
  - [ ] Diameter of access control area
  - [ ] Announcement to university
  - [ ] Announcement to passers-by
  - [ ] Recovery manual

shohei, 2021-03-02:
Model Rocket Safety Code by NAR
https://www.nar.org/safety-information/model-rocket-safety-code/
> Launch Site. I will launch my rocket outdoors, in an open area at least as large as shown in the accompanying table, and in safe weather conditions with wind speeds no greater than 20 miles per hour. I will ensure that there is no dry grass close to the launch pad, and that the launch site does not present risk of grass fires.

Launch site dimensions
Installed Total Impulse (N-sec) | Equivalent Motor Type | Minimum Site Dimensions (ft.)
-- | -- | --
0.00–1.25 | 1/4A, 1/2A | 50
1.26–2.50 | A | 100
2.51–5.00 | B | 200
5.01–10.00 | C | 400
10.01–20.00 | D | 500
20.01–40.00 | E | 1,000
40.01–80.00 | F | 1,000
80.01–160.00 | G | 1,000
160.01–320.00 | Two Gs | 1,500

shohei, 2021-03-02:
>  in safe weather conditions with wind speeds no greater than 20 miles per hour. 

> safe distance of at least 15 feet away when I launch rockets with D motors or smaller, and 30 feet when I launch larger rockets

JeffMboya, 2021-04-21:
Launch sequence documented. Find it in the propulsion repo.

JeffMboya, 2021-04-21:
Here's the link: https://github.com/nakujaproject/Propulsion/tree/master/Launch%20sequence

closed 2021-04-21 by JeffMboya