nakujaproject/internship2021 #7: Finishing N-2 casing fabrication

shohei, 2021-03-01:


shohei, 2021-03-01:
> @Tim and @Jeff Mboya have just finished drilling holes on the casing cylinder body. Tomorrow we'll finish on drilling and tapping on the nozzle and bulk head
[image]

SammyOina, 2021-03-29:
O-rings for n2 nozzle and bulkhead [image]

SammyOina, 2021-03-30:
[image]
O-rings groves have been machined and the motor has been assembled making the motor ready for static firing

closed 2021-03-30 by SammyOina