nakujaproject/internship2021 #27: Ground station monitoring systems (software interface)

willytounsi, 2021-03-04:
Software used in rocket launch monitoring
- Open-source software
- Non-open-source software
Ease of interface with the microcontroller of choice
- Programming language
Others

willytounsi, 2021-03-05:
*******WORK DONE SO FAR*********

1.	LabVIEW (Laboratory Virtual Instrument Engineering Workbench)
-	Visual programming language 
-	System-design platform and development environment that was aimed at enabling all forms of system to be developed.
-	Permits the development of monitoring and control interface with a wide range of sensors and actuators, and microcontrollers, in a short period of time due to its intuitive graphical programming style. 
-	Easy to adapt interface features giving flexibility with the information we will need to display
-	Use by quite a number of rockets launching organizations such as SpaceX and NASA as well as other universities undertaking rocket projects
1.1	Key Features:
	LabVIEW environment:   Consists of:
-	LabVIEW VI manager (project explorer), 
-	The programming tools, 
-	Debugging features, 
-	Templates and ready built sample examples, 
-	An easy interface to hardware drivers. 
	LabVIEW VIs: (“Virtual Instrument”)
-	 Enables a user interface to be built and it contains the programming code.  
	LabVIEW G programming:   
-	This is the graphical programming language where the functional algorithms are built using “drag and drop” techniques. 
	LabVIEW dataflow :   Core concept that determines the running order for the programme

1.2	Advantages of LabVIEW 
•	The graphical interface is flexible and simple to use.  
•	LabVIEW provides a universal platform for numerous applications in diverse fields.
•	LabVIEW can be used with 3rd party hardware: it can be interfaced with C/C++, VB, Fortran etc etc.
•	Easy to interface to many hardware items like data acquisition and test equipment products.
1.3	Availability
-	Commercial version
-	Student version
-	“Cracked version”

2.	Processing  
-	Open-source
-	Graphic library and integrated development environment (IDE)
-	Use for visual design communities
-	Uses java language of programming 

This research is still ongoing and upon completion, I will recommend the software to be used for our ground station.

0x6f736f646f, 2021-04-14:
@willytounsi and @Mike-Kimani  I this is N2 task. Since that is when we will be having telemetry