nakujaproject/internship2021 #34: Design of N-1 rocket including determining which material is to be used to hold the cast propellant

shohei, 2021-03-05:
Ref: https://github.com/nakujaproject/intern/blob/main/felix/Nakuja%20internship%20Progress%20report%20format.pdf

SammyOina, 2021-04-07:
Pipe casing  (PVC) material comparison size 1.5''
Type | Schedule 40 | Schedule 80 
--- | --- | --- 
Inner Diameter (inches)| 1.592 | 1.476
Min Wall Thickness (inches)| 0.145 | 0.200
Max Working Pressure (psi)| 330 | 470
from https://www.jacobsrocketry.com/aer/misc_tables.htm

SammyOina, 2021-04-13:
Pipe used for the casing
Material | UPVC
--- | ---
image | [image]
Pressure Nominal | 25
Outside Diameter | 40 mm
Inside Diameter | 31 mm
Wall thickness | 4.5mm
Estimated Design Pressure | 4.6MPa with a safety factor of 1.5
Burst Pressure | 12MPa with a d-safety factor of 2.62

closed 2021-04-14 by 1998-felix