nakujaproject/internship2021 #35: Simulation of flight performance of N-1 rocket motor propellant

shohei, 2021-03-05:
Ref: https://github.com/nakujaproject/intern/blob/main/felix/Nakuja%20internship%20Progress%20report%20format.pdf

1998-felix, 2021-03-16:
@0x6f736f646f  Please find attached a link to a motor .eng file please add it to open motor and present simulation data. https://github.com/nakujaproject/felixResearch/blob/main/Motor%20data/Motor150321-01/motor-36mm.eng

shohei, 2021-03-16:
@1998-felix I have a question. How did you calculate the .eng data?

1998-felix, 2021-03-17:
@shohei I used openMotor to generate the thrust curves as well as the mass of propellant. I then calculated the mass of PPR pipe of length 90 mm as well as the weight of the nozzle. By inputting these parameters into openMotor I was able o generate a .eng file

shohei, 2021-03-17:
@1998-felix Thank you for the clarification!

SammyOina, 2021-04-08:
# Updated proposed N1-Motor
Grain diameter 40mm
performance | KNDX | KNSU 
--- | --- | ---
Designation | G47 | F46
Impulse | 85.83 | 70.26
Propellant mass (Kg) | 0.18 | 0.14
Average Pressure (MPa) | 1.38 | 1.36
Peak Pressure (MPa) | 1.52 | 1.49
Grain Length (mm) | 90 | 70
Core (mm) | 13 | 13
Nozzle (mm) | 9 | 9
Burn Time (s) | 1.77 | 1.47
Rasp File | [Eng File](https://github.com/nakujaproject/N1-motor/blob/main/Proposed%20Final%20Spec/N1-KNDX-proposed.eng) | [Eng File](https://github.com/nakujaproject/N1-motor/blob/main/Proposed%20Final%20Spec/N1-KNSU-proposed.eng) |
Thrust Curve | [image] | [image] |

SammyOina, 2021-04-08:
> # Updated proposed N1-Motor
> Grain diameter 40mm
>
> performance	KNDX	KNSU
> Designation	G47	F46
> Impulse	85.83	70.26
> Propellant mass (Kg)	0.18	0.14
> Average Pressure (MPa)	1.38	1.36
> Peak Pressure (MPa) [...]

@0x6f736f646f Please find the Rasp file and simulate on openrocket for apogee simulation

0x6f736f646f, 2021-04-09:
> @0x6f736f646f Please find attached a link to a motor .eng file please add it to open Rocket and present simulation data. https://github.com/nakujaproject/felixResearch/blob/main/Motor%20data/Motor150321-01/motor-36mm.eng

Already tested it in openRocket.

0x6f736f646f, 2021-04-09:
> @0x6f736f646f Please find the Rasp file and simulate on openrocket for apogee simulation

It has refused. Please use the format in the test folder. `.eng` should give the specs then `.csv` to give the thrust curve

closed 2021-04-27 by 1998-felix