nakujaproject/internship2021 #48: Research on the BPM 180

shohei, 2021-03-05:
Ref: https://github.com/nakujaproject/intern/blob/main/michael/Nakuja%20internship%201st%20week%20Progress%20report%20.pdf

Mike-Kimani, 2021-03-08:
Done. Please review @shohei 
Find the code here https://github.com/nakujaproject/avionicsv1/tree/main/BMP180

0x6f736f646f, 2021-03-09:
[demo](https://randomnerdtutorials.com/guide-for-bmp180-barometric-sensor-with-arduino/)

shohei, 2021-03-12:
> Done. Please review @shohei
> Find the code here https://github.com/nakujaproject/avionicsv1/tree/main/BMP180

The code seems to be okay. I'd like to see the graph plot to check the error/noise.

shohei, 2021-03-12:
> [demo](https://randomnerdtutorials.com/guide-for-bmp180-barometric-sensor-with-arduino/)

@0x6f736f646f 
Is the demo correct? I found an article on the site, not a demo.

0x6f736f646f, 2021-03-12:
@Mike-Kimani and I tried it but it failed to work

0x6f736f646f, 2021-03-17:
For streaming data from arduino to excel, have a look at [this](https://create.arduino.cc/projecthub/HackingSTEM/stream-data-from-arduino-into-excel-f1bede)

0x6f736f646f, 2021-03-17:
https://www.microsoft.com/en-us/download/details.aspx?id=56976

Mike-Kimani, 2021-03-18:
Thank you @0x6f736f646f

0x6f736f646f, 2021-03-18:
1. [Moving average ](https://maker.pro/arduino/tutorial/how-to-clean-up-noisy-sensor-data-with-a-moving-average-filter) with three window sizes
2. [Exponential average](https://www.megunolink.com/documentation/arduino-libraries/exponential-filter/)
3. [Digital smooth](https://playground.arduino.cc/Main/DigitalSmooth/)
4. [Running median](https://playground.arduino.cc/Main/RunningMedian/)
5. [fir and iir filters](https://github.com/jeroendoggen/Arduino-signal-filtering-library/tree/master/Filter/examples)

Mike-Kimani, 2021-03-18:
Let me do this

0x6f736f646f, 2021-04-06:
Done

closed 2021-04-06 by 0x6f736f646f