nakujaproject/internship2021 #77: Correlate data to find best filter for noisy environment

0x6f736f646f, 2021-03-18:
**Note: for support questions, please use stackoverflow**. This repository's issues are reserved for feature requests and bug reports.

* **I'm submitting a ...**
  - [ ] bug report on the error got from BMP180 and MPU6050 sensor
  - [ ] Analyse which filter has the highest correlation with the real data
  - [ ] Get which filter takes in least memory and requires less computation

* **Do you want to request a *feature* or report a *bug*?**
I want a feature to reduce the amount of noise from the BMP180 and MPU6050 readings

* **What is the current behavior?**
When reading data from the BMP180 and MPU6050 there is fluctuations of values from the real value

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** On running [this](https://github.com/nakujaproject/avionicsv1/tree/main/BMP180) code to you will find the noise in the data

* **What is the expected behavior?**
Steady output values

* **What is the motivation / use case for changing the behavior?**
- Writing a research proposal about it
- Determining accurate apogee and launch

* **Please tell us about your environment:**
  
  - Arduino IDE
  - Board: either stm32, esp32 or arduino mega 
  - Language: arduino, C, C++

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)
None

0x6f736f646f, 2021-03-24:
References
[1.](https://blog.tkjelectronics.dk/2012/09/a-practical-approach-to-kalman-filter-and-how-to-implement-it/)

Mike-Kimani, 2021-05-02:
I think this is done @0x6f736f646f . Close the issue if so

closed 2021-05-03 by 0x6f736f646f