nakujaproject/internship2021 #90: Create "onboarding" manual

shohei, 2021-04-08:
What is Onboarding?
- > Onboarding is a human resources industry term referring to the process of introducing a newly hired employee into an organization. Also known as organizational socialization, onboarding is an important part of helping employees understand their new position and job requirements.
- [https://www.bamboohr.com/hr-glossary/onboarding/](https://www.bamboohr.com/hr-glossary/onboarding/)

Each team (Airframe, Avionics, Propulsion) must have a brief introduction document for newcomers.

0x6f736f646f, 2021-04-11:
For @nakujaproject/airframe-group our doc is [here](https://github.com/nakujaproject/airframeN1/blob/main/Onboarding/newEmployee.md)

shohei, 2021-04-12:
@0x6f736f646f This is great Rodney! Thank you very much.